Antyos/vscode-openscad #1: Highlighting problem

xMase, 2020-03-21:
OS: linux
[image]
what could be the problem?

Antyos, 2020-03-21:
Is the language in the bottom right hand corner set to `OpenSCAD`? If it's an untitled document, it may not auto-detect the language.

xMase, 2020-03-22:
its openscad

Antyos, 2020-03-22:
Well, this is rather embarrassing.... I think I forgot windows isn't case sensitive.... 😂

I think the issue is that the extension calls the grammar file (which is responsible for highlighting text) as `./syntaxes/scad.tmlanguage.json` when in reality, it is `./syntaxes/scad.tmLanguage.json`. 

Before I release a bugfix, would you mind making sure my hunch is correct?

First, navigate to your extensions folder. I believe the path on Linus is: `~/.vscode/extensions`. In the `antyos.openscad-0.1.1` folder, please check that the `./package.json` file is present as well as `./syntaxes/scad.tmLanguage.json` file. In the `package.json` file, under `contributes > grammars`, change line 63: `"path": "./syntaxes/scad.tmlanguage.json"` to `"path": "./syntaxes/scad.tmLanguage.json"` (capitalize the 'L'). 

Let me know if that fixes the issue.

xMase, 2020-03-22:
yes its fixed, thank you!

Antyos, 2020-03-22:
Great! I'm glad everything is working now. I'll try and put out a patch later today or tomorrow.

Antyos, 2020-03-22:
Also, would you be interested in helping me test a feature I'm working on in the `PreviewModel` branch? It enables you to open the current file in OpenSCAD. 

I don't have a Linux machine, and although I could (and probably should) emulate one, it would be nice to get another set of eyes making sure my code works before I put out an official release. If you're interested, we can make another thread for it. If not, no worries.

closed 2020-03-22 by Antyos

xMase, 2020-03-22:
sure but i am not an expert of openscad

Antyos, 2020-03-23:
I just added a new issue for the project, check it out if you're still interested.

I don't consider myself an OpenSCAD expert either, I just got slightly frustrated at the standard editor last September. 😂 I got even more frustrated at the existing VSCode OpenSCAD extensions. A few months later, I have now taught myself `yaml`, `RegEx`, Textmate grammars, `Typescript`, `npm`, `Markdown`, and the VSCode API, all to make this extension for a program I very occasionally use. 😂