Antyos/vscode-openscad #2: Testing "Preview" Command

Antyos, 2020-03-23:
I would like some feedback on the "Preview in OpenSCAD" feature.

Steps (assuming VSCode as IDE):
1. Fork the Repo (or clone if you have no desire to make changes. I'm a Github noob; no hate mail if that's considered bad practice please)
2. Checkout the branch `PreviewModel`
3. Run `npm install` in the directory to get all the modules
4. Run `npm run compile` to compile the extension
5. Go to the debugger and click the play button. It will launch a developer instance of VSCode.
6. In an `OpenSCAD` file, try out the commands: 
    - `OpenSCAD: Preview Model`
    - `OpenSCAD: Kill OpenSCAD Preview`
    - `OpenSCAD: Kill All OpenSCAD Previews`

Make sure you have OpenSCAD installed. Depending on your OS (i.e. not Windows), you may need to go into the extension settings and change the installation path.

Let me know any thoughts and opinions you have on the functionality of these commands. Anything is appreciated!

I will note, I plan to some add features (at the time of writing this)
- Add a button on the editor tab bar to make launching previews easier
- Add a menu in the context menu (right click) of `scad` files to launch a preview of said file

xMase, 2020-03-23:
os: linux Manjaro

- [x] OpenSCAD: Preview Model
- [x] OpenSCAD: Kill OpenSCAD Preview
- [ ] OpenSCAD: Kill All OpenSCAD Previews (**COMMAND NOT FOUND**)

Antyos, 2020-03-24:
Oh, the embarrassment of forgetting to push your changes... 😂

Sorry about having you look at half-finished work, it should be MUCH better now. There are now checks to limit open windows and prevent duplicates of the same file.

I'll see what I can do about F6 to automate rendering. I should be able to add a key binding to preview or even render the model. However, once the instance of OpenSCAD is started, I think it is out of my control for sending more arguments.

You are able to have it render and export a `.stl` or a few other file types through the command line, so I plan to add a command to do that through VSCode.

xMase, 2020-03-24:
now everything seems to be working properly. 
only the syntax is not displayed correctly on linux on this branch.

Antyos, 2020-03-26:
Glad to know everything's working.
Syntax highlighting will be fixed when I merge back into master.

Antyos, 2020-06-21:
I think this "issue" has served its purpose. While feedback from others is always appreciated, I'm not sure this is the best place to do it. Mostly, seeing the one open issue (that I know isn't an issue) is bugging me, so I'm going to close it.

closed 2020-06-21 by Antyos