Antyos/vscode-openscad #4: Can't use openscad-nightly

hjalte33, 2020-09-13:
I tried changeing the path to openscad in the settings so it would point to /usr/bin/openscad-nightly but this addon seems to be hardcoded to a binary called openscad since i get the error "cannot find openscad command"  
are there any way i can point to the nightly build instead?

Antyos, 2020-09-14:
If you set the value of `openscad.launchPath` in your config (so it isn't blank), the extension should use that string exactly when calling OpenSCAD.

Admittedly, the error message **is** hard coded so it only tells you it "can't find the command 'openscad'" instead of the path it actually uses. That will be changed in the next release.

Try launching your nightly build of OpenSCAD via a terminal / command prompt window, then copy whichever command you used to launch it into the configuration `openscad.launchPath`. If the nightly build executable is in the system path, then you should just be able to type the name of the file instead of the whole path. 

If that doesn't work, I'll need a little bit more information, like:

- What OS you are using?
- Does the extension work with a non-nightly build?
- What value of `openscad.launchPath` are you trying in your settings?
- What is the path to the file that launches the nightly build of OpenSCAD?

hjalte33, 2020-09-14:
Thank you this worked as intended. Just a bit confusing that it said "overwrite default path..."

closed 2020-09-14 by hjalte33

Antyos, 2020-09-14:
@hjalte33 I'm glad everything is working. Thanks for bringing this to my attention because that is not what the description is supposed to read. I thought "Override the default path" only read in the contributions section; the description in settings should actually read:

> Command to launch `openscad`. Either the path to the openscad executable, or just "`openscad`" (no quotes) if the executable is in the path. If left blank, it will use the default path for your system noted below:
> - Windows: `C:\\Program Files\\Openscad\\openscad.exe`
>     _Note:_ Must specify path to `openscad.exe` on Windows for kill command to work properly.
> - MacOS: `/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD`
> - Linux: `openscad` (Automatically in path)

If this is what the description said, would that setting's function make more sense?

I will look into why the various configuration descriptions are not showing up as I had intended.