Antyos/vscode-openscad #9: Preview error

Neiromaster, 2020-12-18:
I get an error when trying to run previews on Windows 10 
[image]

Antyos, 2020-12-18:
I don't think the extension uses Boost, so I'm not sure where that's coming from.

If you don't mind giving a bit more information:
- What version of the extension are you running?
- Where do you have OpenSCAD installed?
- What do you have the config `openscad.launchPath` set to?
- What method did you use to open a preview?

Neiromaster, 2020-12-22:
OpenSCAD for VSCode version 1.0.2
I instal the application in the standard path and I launch it with the button at the top of VSCode

[image]
[image]

Antyos, 2020-12-22:
I ran the preview command with both OpenSCAD v2020.12-rc3 (I could not find a stable release for v2020.12) and v2019.05. Both versions were able to preview without issue.

Just going down the list of questions/tests --
- What version of VSCode are you using?
- What build of Windows 10 are you using?
- Have you tried disabling all your other extensions and trying the "preview" command? 
 
There's an option to disable all extensions in the 3 dots context menu at the top of the extensions pane. If it works when it is the only enabled extension, then it may be another extension causing a conflicting issue. Once we know that, we can better solve the problem.

And fyi, you don't need to include the path in the extension configuration if OpenSCAD is installed in the default location, though I appreciate it the clarity for the screenshot.

---

If you want, you can try cloning the repository and running the extension in debug mode -- that might be more telling. Follow the steps in [Contributing](https://github.com/Antyos/vscode-openscad#contributing) to set up the repository. 

To debug it: (I need to update the README to include this)
- Open the local repo in VSCode
- Rename `.vscode/launch-template.json` to `.vscode/launch.json` to enable the launch configuration
- Running `Extension` in the debug pane will open a new window of VSCode with the extension running
- Try running the preview command in the debug window and check the debug console in the other window to see if there are any messages

I hope this can be of some help! I wish I knew what was happening.