Antyos/vscode-openscad #10: Publish on Open VSX

tiramiseb, 2021-01-04:
As I prefer true open source software, I am using VSCodium. Thus, I don't have access to the Microsoft VSCode store thing.

However, there is an open source alternative for this too: Open VSX.

The process to publish on that registry is detailed here:
https://github.com/eclipse/openvsx/wiki/Publishing-Extensions

Could you please do that?

Thanks!

Antyos, 2021-01-07:
I've been meaning to publish on OpenVSX for a while. Thanks for the motivation to finally do it.

The extension should be available now. Here's a link: https://open-vsx.org/extension/Antyos/openscad

Let me know if there are any other issues!

closed 2021-01-07 by Antyos

tiramiseb, 2021-01-07:
It's working great, thanks !