Antyos/vscode-openscad #16: OpenSCAD Language Server

Tomatower, 2021-02-07:
I have started implementing the Language Server Protocol interface for OpenSCAD, and I would love to see it used in the already best vscode extension for openscad: https://github.com/openscad/openscad/pull/3635
The Idea is, that we can start `openscad --lsp-listen 23725`, connect to port 23725, and vscode-languageclient takes care of all the fancyness.

I have created a super minimal extension for the basic support, which will grow with the openscad interface (for example "hover", autocomplete with defined module names, ...). I have already planned to add a bunch of features.
You can find my extension here: https://github.com/Tomatower/vscode-openscad

The biggest change to this extension would be `openscad.render` will change, and some logic that has to check if the currently running version does support LSP.

Do you want to have a PR for this upgrade?

Antyos, 2021-02-08:
> Do you want to have a PR for this upgrade?

***Absolutely. I am totally on board with this.***

I have been wanting to implement a language server for a while, but haven't had the chance. I am absolutely delighted to see someone asking about wanting to add one as a PR! Let's make this work!

I do have one question/concern--with [openscad/opendscad#3635](https://github.com/openscad/openscad/pull/3635), does that mean the language server will only work in a future version of OpenSCAD (assuming the feature is merged)? Or, will the language server still be able to work in older versions of OpenSCAD. My main hesitation is that this feature won't be available for months or only for users of experimental builds, though I suppose that's more a matter of patience than anything else.

Also, how are you planning to work on the language server? I mainly ask because your extension repo is not a fork of this one. Are you planning on creating it for this extension or are you creating a general language server extension? If it's the former, I think it would be easiest to develop the server on a fork of this codebase. If not, we can find some way to bundle them together. There are pros and cons to both options.

Tomatower, 2021-02-08:
It is very likely to get merged, in the next couple of weeks.

Unfortunately, this is exactly what that means. The Languageserver is integrated into OpenSCAD, enabling us to trigger previews from vscode and whatever random Ideas we come up with (related to #6, but also on non-dbus systems.)

The language server will be only in the experimental builds, and it will take a few months until the next release of openscad (the last one was a week ago.)
But the latest experimental version can be packaged with this extension or download it on request.

My "extension" is only like ~90 lines of code, which I will integrate as a fork from this repo. (see https://github.com/Tomatower/vscode-openscad/blob/master/client/src/extension.ts - the rest is the remainder of the `lsp-sample` project)

Antyos, 2021-02-09:
Alright, sounds good! 

If we want to bundle a preview version with the extension, I'll need to change the license from LGPL3 to GPL3 (which I've been meaning to do for some time now).

PR #17 (2021-02-09): added initial languageserver connection
Let me begin implementing #16 (althought it is not yet merged upstream...)
This is my first - and last - typescript/vscode extension I will ever write, so any help is greatly appreciated!

- [X] Add connection to a running OpenSCAD instance
   - [X] Receive Diagnostics
   - [ ] Trigger a preview in the linked instance (registered as command `openscad.lsp.preview`)
- [ ] Query the installed openscad version if it does/does not support the LSP magic (check if it starts with command line flag `--lsp-listen`, which is the switch for the languageserver)
- [ ] [... cut]
closed 2021-02-09 by Antyos