Antyos/vscode-openscad #21: Adding a formatter

matthew-viglione, 2021-05-13:
I recently realized this wonderful extension does not include a formatter. After looking around I found there aren't really any good ones out there and this would great to add. I'll see if I can do it in my spare time, but maybe in the meantime someone else knows of one or a workaround? (Maybe using a formatter for another similar language?)

Antyos, 2021-05-19:
Sorry for the late response!

I am not aware of any OpenSCAD formatters, but I'd love to have one in the extension! 

While I am sure there are some formatters that could be used as a starting point, like perhaps one used by a C-like language or maybe even JavaScript, I don't think it could be used in its raw form. OpenSCAD has different syntax for things like the `module` keyword, as well as several unique language characteristics. 

For example, how "should" the following code be formatted?

```scad
translate([0, 0, 10]) rotate([90, 0, 0]) translate([10, 0, 0]) cylinder(r=5, d=10);

// Or

translate([0, 0, 10])
  rotate([90, 0, 0])
  translate([10, 0, 0])
  cylinder(r=5, d=10);

// Or

translate([0, 0, 10])
  rotate([90, 0, 0])
    translate([10, 0, 0])
      cylinder(r=5, d=10);
```

I'm sure there are plenty of other things I'm not thinking of too. That is not to say I am not on board with this idea, I would absolutely love to have a formatted, I just think it's worth talking about the implementation or formatting guidelines to be implemented.

mauriceoegerli, 2021-07-10:
There is https://github.com/Maxattax97/openscad-format, which uses ClangFormat. Or maybe you have already seen that.

matthew-viglione, 2021-07-16:
@mauriceoegerli I haven't seen this before, I'll go give it a try.