Antyos/vscode-openscad #27: Extension causes high cpu load

mrspaceman, 2021-10-15:
- Issue Type: `Performance`
- Extension Name: `openscad`
- Extension Version: `1.1.1`
- OS Version: `Linux x64 5.11.0-37-generic`
- VS Code version: `1.62.0-insider`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`/tmp/Antyos.openscad-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

mrspaceman, 2021-10-15:
CPU Profile :
[Antyos.openscad-unresponsive.cpuprofile.txt](https://github.com/Antyos/vscode-openscad/files/7350924/Antyos.openscad-unresponsive.cpuprofile.txt)

Antyos, 2021-12-22:
I finally have some time to investigate the high CPU load. Is this still an issue @mrspaceman?