Antyos/vscode-openscad #28: Hover Information

IGLOU-EU, 2021-12-10:
Hello,
first of all, thank you for this add-on, it's great!

I'm learning to use OpenSCAD, and it's not easy ...
A function like "Hover Info" would be a great help.
Like this, for example. Give function info, arg ...
[image]

Would it be complex to add? Could I be of any help for it?

Thanks again :)

Antyos, 2021-12-13:
Thank you for the kind words! I would love to add hovers to the extension; I can't tell you how many times I've been writing some OpenSCAD in VS Code, attempted to launch hovers/autocompletion, and got annoyed at the developer for not having support for them, until realizing that *I* am the developer. 😂

From the VS Code API, hovers are included under [Programmatic Language Features](https://code.visualstudio.com/api/language-extensions/programmatic-language-features). It looks like there are two ways to implement hovers: by creating a hover provider directly and by creating a Language Server. 

Although it is more work, I'd prefer to go down the language server path because it will be more versatile down the line. I know @Tomatower had worked on implementing some basic Language Server stuff a few months ago in #16 and #17, but I do not know what is happening with that now. If the language server is fully working, I think we should be able to create a hover provider with relative ease(?).

I'm not sure when I'll get a chance to fully investigate the language server and hovers. If you want to look into the Language Server that we currently have ([src/languageclient.ts](https://github.com/Antyos/vscode-openscad/blob/Tomatower/languageserver/src/languageclient.ts)), what we would need to integrate a Hover Provider, and create a PR, I would be more than happy to look over some code!

Tomatower, 2021-12-13:
Unfortunately the language server has still not been merged upstream into openscad (see https://github.com/openscad/openscad/pull/3635) 

And it does not yet support introspection into the parsing tree, but it should be possible (hover support with variable value parsing is one of my most missing features of openscad...)

I cannot predict when I will have the time to further dive into this.

Antyos, 2021-12-23:
I am currently working on bringing this extension to the VS Code web editor as a [Web Extension](https://code.visualstudio.com/api/extension-guides/web-extensions). Features that require an installed OpenSCAD client, such as previewing a file, are not likely to work online, but at the very least, I am hoping to get base language features, like syntax highlighting working.

Even if the OpenSCAD language server is merged upstream, I don't think the online editor would be able to access it. Perhaps we could bundle the language server in the extension somehow? That way we can bypass the wait on OpenSCAD's end, and it would work online.

If this is something that sounds like it has potential (of course, with a loose time frame here), I can open a new issue for that discussion.

Lenbok, 2022-03-12:
In case it helps, there is tree sitter support for openscad, at https://github.com/bollian/tree-sitter-openscad as well as an alternative language server that uses it, at https://github.com/dzhu/openscad-language-server (it doesn't require openscad itself).