Antyos/vscode-openscad #29: highlighting issue #2

Hr46ph, 2022-01-08:
Hi there!

First of all, thanks for your work on this, the integrated openscad editor is too barebones and annoying to use.

I use vscode for everything I can use it for. Im on the free/libre version on Arch Linux (not the MS version) if that makes a difference (I don't think so).

Highlighting doesn't seem to work. i found the other thread about it:
https://github.com/Antyos/vscode-openscad/issues/5

I followed the tips and I can see the package.json contains the correct name and it shows OpenSCAD in the bottom right hand.

I get a few things highlighted, comments are green and text between double quotes is red. Also numbers integers/numbers are green (slightly different shade from comments).

I checked the extension settings and code highlighting is set to auto. I can set it to 'original' but doesn't make a difference.

Any ideas please?

Thanks!

Hr46ph, 2022-01-08:
Well fast solution from myself, highlighting works when selecting dark+ theme. I was on Light (non-plus).

I'll close the issue.

closed 2022-01-08 by Hr46ph

Antyos, 2022-01-09:
Well, I always like to see issues that sort themselves out before I can even get a chance to look at them ;)

I'm currently working on a big update (adding web compatibility), so I'll try to add something in the README about making sure you use the correct type of theme, i.e., **Dark+ (default dark)** vs **Dark (Visual Studio)**.

Hr46ph, 2022-01-09:
Awesome!

Thanks for your work on this, appreciate it.