Antyos/vscode-openscad #30: TODO: Housekeeping

elasticdotventures, 2022-05-02:
Hi @Antyos 

I'm putting together:
https://github.com/elasticdotventures/awesome-openscad

I hope to do a bit of community TIDY, curation and housekeeping stuff/proposals today on the vs-code marketplace related to openscad plugins. 

I'd like to be a regular contributor here *possibly* adding my own code, but initially beginning with some build tests & badges, tests and other indicators that this plugin is  accepting contributions, and writing a RUST cargo packaging interface for Linux users, some other stuff on my list. 

For openSCAD import:
I'm additionally contemplating a new project/addition about building a cross platform openscad browser/marketplace tooling LSP-extension 'checkbox' optional-feature library manager that can be used by many (*any) OpenSCAD visualizer as a way to quickly find and install objects & tools (using RUST cargo).  

For export:
I'm going to try and figure out how to do it with other plugins so that (for example) the git lineage aspects happen correctly in the background,  in an opinionated manner with lots of interfaces to other software, especially with regard to external simulators such as blender, so for example the git build actions are tied to the object validation onChange in vscode.  

I'd like to see this LSP project integrate with the OpenSCAD blender, and be used inside blender (which I think, blender already supports other LSP, so hopefully not too much work) .. i.e. LSP is not specific to vs-code and integrate this plugin into https://github.com/elasticdotventures/Blender-openSCAD
as a bundle in [elasticdotventures/awesome-openscad]()

I think openscad has a ton of utility, especially if more complex & standardized idiomatic compositional interfaces can be built and shared across projects.  Presently imho the browser experience sucks and requires a lot of toil by each developers to curate their own openSCAD tooling & object browser-discovery interface.

elasticdotventures, 2022-05-02:
when I say openSCAD blender, I mean this LSP project shouldn't be limited to strictly vscode, it should be possible to have it work in other editors (for openSCAD there are MANY) -- the advantage of bringing LSP's to the various openscad editors such as cadhub, jsopenscad etc.  I hope that is clear, LSP is not specific to VSCode.

Antyos, 2022-05-12:
Hey! I'm not sure how I missed this earlier--oops!

I like the sound of what you are trying to do. If I understand correctly, you are trying to make a unified database of OpenSCAD utilities, including:
- Information about plugins/extensions/editors that can be used for OpenSCAD, are actively accepting contributions, etc.
- A package manager for various OpenSCAD libraries
- A universal LSP that can be used in any editor

What specifically would you have me do to help with this project? I know you mentioned adding badges, are there any ones in particular I should add?

---

One of my long-term goals with this project is to add a custom "export as stl" syntax, which would include the ability to add a comment to the top of a `.scad` file such as:

```scad
/* sample.scad */

//export:${fileNameNoExt}_${var:diameter}mm.${exportExt}

diameter = 5;

cylinder(d=diameter, h=10);
```

Which would export a file `sample_5mm.stl`.

It may be worth creating an issue page to discuss this syntax at some point. Would having something like this implemented be of interest to your project?