Antyos/vscode-openscad #32: Query and feature request musings

Dorro101, 2022-06-26:
Hi Antyos

Thanks so much for this great extension, I find this extension invaluable for developing my prototypes for my Ender 3/procedural 3D models for my game, especially as my OpenSCAD scripts have become exponentially complex, modular and multi-file. I have been consolidating my editing in VSCodium for my writing (Markdown, Zen mode, spelling and grammar checking and Github integration are just a few of the great features available), OpenSCAD, Blender and Godot scripting on Linux (Fedora 36). It is working pretty well because of all the wonderful extensions like yours that are freely available.

I have just started getting to grips with debugging in VSCodium (Godot and BPython). I was wondering if is integrated for OpenSCAD is planned. I happened on this as I would love to have in-line values for variables for OpenSCAD in the editor, which would help as a visual reference e.g. something that looks like this: `dongle_size = [dongle_cover_widthX, dongle_cover_heightY + tolerance]; // dongle_size = [32, 8, 0.5]` and seems to be a thing is some integrated debuggers. I only just started learning to program, so my current skills are dangerous at the best of times...

Just curious, so apologies for the random post as I am not sure where else to message you here. So feel free to close this thread if you consider it spurious.

Ciao
Dorian

Antyos, 2022-07-05:
Hi Dorro! Thanks for the feedback and congrats on getting into debugging tools!

As of right now, I am not sure if I fully understand your question. Are you wanting a runtime debugger with breakpoints that allow you to pause the execution of a file to inspect the state of local/global variables? Or are you wanting some sort of overlay that shows the numerical readouts of each value?

If it's the former, I am not sure if that is possible due to the way OpenSCAD handles variables. Unlike most languages, OpenSCAD [sets variables at compile time, not runtime](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Variables_are_set_at_compile-time,_not_run-time). Take this example from the wiki:

> ```openscad
> // The value of 'a' reflects only the last set value
>   a = 0;
>   echo(a);  // 5
>   a = 3;
>   echo(a);  // 5
>   a = 5;
> ```
> While this appears to be counter-intuitive, it allows you to do some interesting things: for instance, if you set up your shared library files to have default values defined as variables at their root level, when you include that file in your own code you can 're-define' or override those constants by simply assigning a new value to them. So changing constant values gives you more flexibility. If constants would never change, of course, you always can be sure having the value you see in any constant definition. Not so here. If you see a constant value definition at any other place its value could be different. This is very flexible.

While I may personally disagree with this design decision, it is part of the language. (In my opinion, libraries should handle changeable constants using parameters with default arguments, not by allowing mutation of global constants.)

If you are asking for the latter (a numerical readout overlay), that is still a bit complicated, but is probably doable.

If it helps in the mean time, you can use [modifier characters](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters) and/or a lot of [`echo()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Echo_Statements) for debugging.

Dorro101, 2022-07-06:
Hi Antyos

Thank you for taking your time to respond so thoughtfully, much appreciated. I think my lack of experience may have me at a disadvantage here. I was hoping to see the live numerical values next to the variables (as an in-line comment for example) in the code view in VSCodium, as I am better able to visualise things in my head if I can see the actual values alongside the more abstract variable names, as my mental maths sucks. I have been using the modifiers and echo a lot. I think I understand the variable thing and is useful for my included files, although I find myself changing the way I structure my code in order to reduce redundancy and find myself wasting a lot of time re-factoring code instead of working on my prototypes. A preview window in VSCodium would be nice, but I have set VSCodium to auto-save when changing windows and just switch to OpenSCAD and back again.

Sadly, I have found OpenSCAD's poor performance while previewing and exporting STLs interrupting my work flow as my projects become more complex, especially when checking my combined parts for fit, issues and design before slicing in Cura...

Thanks again for your time and the time and effort you spend on this great extension.
Ciao
Dorian

Antyos, 2022-07-09:
I like the idea of inline hints showing the value of variables. It would be a non-trivial amount of effort to implement, but I think it would be useful to put on the "someday" features.

There are two parts to this:
1. The overlays themselves should be relatively straight forward to implement. The VSCode Decorator API can be used for this. See: <https://vscode.rocks/decorations> and <https://github.com/microsoft/vscode-extension-samples/blob/main/decorator-sample/USAGE.md>
2. Getting the actual variables. This is the hard part because it requires a parser/interpreter. It is possible this will be easier with the implementation of a language server in #31, but I am not too familiar with all of that.

I am going to close this issue and open a new one with more detail about the specific feature being requested.

---

As for OpenSCAD being laggy, that is also one of my biggest gripes with the software. Knowing that software like Blender (written in Python) can handle millions of vertices without issue, it is an ever-present mystery why OpenSCAD (written in C++) struggles to handle a few holes. Check out this note in the FAQ about [why OpenSCAD previews are so slow](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_the_preview_so_slow?). You may also want to check out the [`render()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Render) function which forces the generation of a mesh even in preview mode. It sounds like it may help with performance.

closed 2022-07-09 by Antyos