Antyos/vscode-openscad #34: can not use in vscode

ghohoj, 2022-09-13:
system:windows11,OpenSCAD 2021.01 Windows
It reports:
Cannot find the command: "E:\openscad\openscad.exe". Make sure OpenSCAD is installed. You may need to specify the installation path under `Settings > OpenSCAD > Launch Path`
however,the path is right,I can open it in cmd
and I try to use openscad try.scad -o try.png
It's OK.
please help me.

Antyos, 2022-09-14:
Hi ghohoj, I'm not quite sure what is wrong, but I will do my best to help. I have a few questions:

- Which command are you running that causes this?
- What do you have the configuration `openscad.launchPath` set to? I assume `"E:\openscad\openscad.exe"`, but I want to double check.
- What happens if you run `E:\openscad\openscad.exe try.scad -o try.png` in a terminal?

ghohoj, 2022-09-14:
Which command are you running that causes this?
It's my first time I use openscad in vscode.I try to click `preview in openscad`,and it tell me that my launchPath is wrong.

---

What do you have the configuration set to? I assume , but I want to double check.`openscad.launchPath"E:\openscad\openscad.exe"`
Yes,this is the only thing I change in the configuration. the other configuration is default.

---

What happens if you run in a terminal?
I will get a png picture named try.png

Antyos, 2022-09-14:
Hmm. There are two things I can think of:

1. Either VS Code doesn't have permission to access the `E:` drive.
2. The regex that drives the library used to check if the `openscad` command is a valid path is failing for an absolute path to the `E:` drive (which I thought I fixed a few years ago).

Try running VS Code as an administrator. If that doesn't work, we'll figure something else out.

ghohoj, 2022-09-15:
I have tried to do it, but it failed. And I try to download openscad in C: drive, it didn't work.

Antyos, 2022-09-16:
I have no idea why it won't work, so we're going to have to try a few things.

Can you follow the steps to the Contributing section of the README to run the extension in debug mode? The README assumes you already have Node installed (I need to update it to include a note about that).

These lines prevent VS Code from launching an instance of OpenSCAD if if the path to `openscad.exe` is invalid: https://github.com/Antyos/vscode-openscad/blob/master/src/preview.ts#L120-L124. Please comment them out. I want to see if the issue is with the detection of the executable, or if VS Code is genuinely not able to access it.

Please let me know what happens when you try to preview a file while running the extension in debug mode.

ghohoj, 2022-09-21:
I have given up. Now I use vscode+cmd, it can achieve the same effect. I write 
`{
    "code-runner.executorMap": {
        "scad": "cd $dir && openscad --preview utf-8 $fileName"
    },
    "code-runner.runInTerminal": true
}`
in `setting.json`. And when I click run, I can preview the result.

Antyos, 2022-09-21:
I'm sorry I could not be more help, but I am glad you were able to get something working! I will go ahead and close this issue for now--if further developments are made, we can always re-open the issue.