Antyos/vscode-openscad #36: Using the preview feature with flatpak

Poly2it, 2022-11-19:
I have installed OpenSCAD using flatpak, and it seems that vscode-openscad doesn't recognize my launch command:
`flatpak run org.openscad.OpenSCAD`.

Error message:
```
Cannot find the command: "flatpak run org.openscad.OpenSCAD". 
Make sure OpenSCAD is installed. 
You may need to specify the installation path under `Settings > OpenSCAD > Launch Path`
```

Antyos, 2022-11-28:
Sorry, forgot to respond to this earlier.

The current way the extension checks for a valid launch command is by checking if it is a file or not. This was an effective solution before I realized something like Flatpak might exist. A few thoughts I have to fix it:
1. Hard-coding all the different package managers OpenSCAD may be installed through. I'm not a big fan of this because while it isn't _necessarily_ an issue to do, it does make maintenance more challenging.
2. Try calling the specified command and only show an error if the process fails to start
3. Call `{openscad.launchPath} --version` to check that it is both a valid 
4. Other ideas are appreciated.

I will look into this when I have some more time once my classes slow down a bit.

In the mean time, you could try writing a wrapper script that just passes everything `flatpak run org.openscad.OpenSCAD`. That way, the extension can detect that a launch file exists.

Antyos, 2023-03-19:
This should be fixed by #43. I have added a new configuration for `openscad.launchArgs`. You should be able to use Flatpak with the settings:
```json
{
    "openscad.launchPath": "flatpak",
    "openscad.launchArgs": ["run", "org.openscad.OpenSCAD"]
}
```

closed 2023-03-19 by Antyos