Antyos/vscode-openscad #38: Cheatsheet Enhancement

RAMilewski, 2023-01-19:
Would it be possible to add a second cheatsheet link that is user configurable?

Having the BOSL2 cheatsheet accessible in same way as the OpenSCAD cheatsheet would be an enormous convenience.   

Thank you for making VSCode the best OpenSCAD editor on the planet!

Antyos, 2023-02-10:
Hey! Thanks for the kind words! Sorry it's taken me so long to get back to you about this--

Since BOSL2 seems to be a popular library, I could probably hard-code it in. That being said, I will think if there is a good way of adding user-defined cheatsheets.

I think having a proper language server implemented would also help your needs. It's been on my to-do list for far too long now and it still hasn't been done. I'll do my best to work on this soon, but I've been very busy with my classes.

RAMilewski, 2023-02-16:
Hard coded would be great if that's a lot easier.  I wasn't sure if the rest of the world was as enamored of BOSL2 as I am.  I don't even think about using OpenSCAD without an include<BOSL2/std.scad> at the top.

User-defined cheatsheets would be more universal though.

Good luck with your classes!   Never stop learning!

Antyos, 2023-03-18:
Hey! So I'm (slowly) making progress on things. I checked out BOSL2 and it looks really cool (I could see myself using it at some point).

The [BOSL2 cheatsheet](https://github.com/revarbat/BOSL2/wiki/CheatSheet) is massive and does not seem to be in a format that I could easily stick into the extension. This may be what you had initially suggested, but here is what I'm thinking: 

I can add a configuration to a user-specified HTML file for a second cheatsheet that can be launched via alt-clicking the "Open Cheatsheet" button on the status bar.

Thoughts?

RAMilewski, 2023-03-19:
That sounds like a great solution!  I'll file an issue in the BOSL2 repository and see what the guys say.  It may be that something like the function index would make more sense than the cheatsheet.   ...or maybe just that right sidebar from the BOSL2 wiki.  I don't know if that's directly linkable, but I'm sure it wouldn't take much work to make it so.

Thank you for your attention to this!

RAMilewski, 2023-03-19:
...and once you start using BOSL2 you'll never go back to naked OpenSCAD.

RAMilewski, 2023-03-20:
I think the [Table of Contents](https://github.com/revarbat/BOSL2/wiki/TOC) makes the most sense for now.  When opened in a narrow window as VSCode does, it pushes the sidebar to the bottom, so I suspect it might work.

Keeping the url user-definable still seems like a good idea for folks who may have other preferences.