Antyos/vscode-openscad #39: Adding a linter

MarkJeronimus, 2023-01-25:
I'm not a high-duty OpenSCAD user, but one of the things that it's editor lacks and I was looking for in a replacement, is error highlighting, in particular syntax errors and unknown variables/identifiers.

Antyos, 2023-02-10:
Hi Mark, I think this is something that will be resolved by (or at least requires) the language server introduced in #31. The people who have worked on it previously seem to have moved on, and I haven't had the time to look into myself, so I don't know when that will happen. I haven't forgotten about this, though!

julianstirling, 2025-04-23:
If you wanted a run-on-save linter I wrote SCA2D. 
https://gitlab.com/bath_open_instrumentation_group/sca2d

It will probably annoy a lot of people as it warns about a lot of language features that we think are unsafe but people use all the time. (globals, scope renaming, `include`), it was written for the OpenFlexure project to help us maintain a very large OpenSCAD codebase.