Antyos/vscode-openscad #50: Launch Path Problems

RAMilewski, 2023-08-21:
1.2.1 MacOS version stopped finding OpenSCAD

After the update I get the message: 

> Cannot find the command: "". Make sure OpenSCAD is installed. You may need to specify the installation path under `Settings > OpenSCAD > Launch Path`

OpenSCAD.app is in the same location it has always been, which matches the default:

> /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD

I tried explicitly adding that path but to no avail.

Antyos, 2023-08-24:
Did it work in version 1.1? It probably has something to do with the new way I changed the extension to detect the executable in #46.

Are you able to run `openscad --version` in the command line? What does it output?

mkkoch, 2023-08-28:
``/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD --version
OpenSCAD version 2023.08.21``

Antyos, 2023-08-31:
I don't have easy access to a Mac, so I can't test things this instant. In the meantime, do you know if it works with older versions of the extension and/or OpenSCAD?

RAMilewski, 2023-09-18:
Works fine with extension v 1.1.1 and OpenSCAD 2021.01 or 2023.09.15
Fails with extension v1.2.0 or v1.2.1 with either version of OpenSCAD.

Antyos, 2023-10-09:
I have just released v1.2.2 of the extension. It includes a fix that may have resolved your issue, but it also includes additional logging information. Can you give it a try and let me know if the issue is resolved?

RAMilewski, 2023-10-09:
v1.2.2 works for me with OpenSCAD 2023.10.08 and 2021.01.

Thanks for the fix!

closed 2023-10-09 by Antyos