Antyos/vscode-openscad #56: [FEATURE REQUEST] Auto-launch slicer after export

leo-nightfever, 2024-01-02:
My workflow is always the same:

1. Edit the model in VS Code
2. Preview the model in OpenSCAD
3. Export an STL via VS Code plugin
4. Open the model in Cura and slice it
5. Upload G-code to printer and start printing

The only thing that is not yet linked together are steps 3 and 4. I still need to save the STL and open it manually via Explorer. Would be great to have the option to have an _"'Export Finished'-Action"_ where I could specify a path to a program that automatically gets called after an export is finished, together with the path to the just exported .stl. (much like the preview button in VSC triggers OpenSCAD with the currently opened .scad file)

Antyos, 2024-01-02:
This is a really cool feature idea, I like it. I know there are some extensions (it may be built-in now) that implement "code actions on save", so I don't think it would be _too_ difficult to add this. Ideally, I'd like to implement this as a callable VS Code task after the export action is run, but I may end up just having it call something in the command line.

As I mentioned in your other issue #55, I'm not sure when I'll be able to get around to this (this one is a bit more involved than the export name formatting). I'd be happy to accept a PR, though!

leo-nightfever, 2024-01-02:
Sadly I'm not really experienced enough in coding these plugins to really help out a lot. I might look into the date thing to see if I'm able to figure anything out.

rizwansarwar, 2024-03-10:
Could be cool, when generating STL, trigger an external script with path to STL. I don't know much about VSCode build system but this could possibly be one route.