Antyos/vscode-openscad #62: OpenSCAD launch is not working if executable differs from openscad.exe

snlitotes, 2024-04-26:
__Problem__
If I specify the launch path to e.g.:
`"openscad.launchPath": "d:\PortableApps\PortableApps\OpenSCADPortable\OpenSCADPortable.exe"`
then the launcher exits with an error message:
~~~
Cannot find the command: "d:\PortableApps\PortableApps\OpenSCADPortable\OpenSCADPortable.exe". Make sure OpenSCAD is installed. You may need to specify the installation path under `Settings > OpenSCAD > Launch Path`
~~~

__Background__
To run OpenSCAD in a portable environment there is a wrapper executable needed that must have a different name from openscad.exe. So it is needed that the plugin-launcher is capable to run a launch path that can have an arbitrary name.

snlitotes, 2024-04-26:
The reason might be that the OpenSCADportable wrapper forwards the parameter to the openscad.exe but does not print any values to the console.

Antyos, 2024-04-26:
I've never heard of PortableApps before--I don't see OpenSCAD listed in their apps (https://portableapps.com/apps), where did you download it from? Not being able to forward command line output in a wrapper seems like a strange oversight. I wonder if it's a potential issue with the way it was generated.

Regardless, this may be a sign that I should create an "advanced" option to skip the version check. The previous way of checking the validity of the openscad exe was to check if the file itself existed, but this caused issues for use with Flatpak (#36). I don't really like "skip safety checks" as a principle, but I'm not sure how else to resolve this issue. Suggestions are welcome.

snlitotes, 2024-04-26:
You see it not listed because nobody ported it to portableapps.com yet. This was my intent. My problem is that the portableapps launcher (based on NSIS) does not pass return values back to the console. I already had a hard to find out how to pass console messages back with NSIS. But the concept of it just did not forsee it.
Skipping the version check would be an option. And I fully agree - just skipping safety checks is ugly. I would do that neither. If you like you can leave this issue open until somebody else comes with the same request or you just close it as not planned. Both is fine for me.

fenhl, 2024-06-05:
I installed OpenSCAD using [Scoop](https://scoop.sh/) and I'm having the same issue — its shim for the `openscad` command also doesn't output anything when running `openscad --version`.

Antyos, 2024-06-06:
I really wish these wrappers would respect command line output, but there's no point in complaining about something far beyond my control.

Maybe the solution should then be to check if the command runs successfully rather than listening for specific output. If there is no output to `openscad --version`, then it sets the version to `undefined`, but still continues as normal. In principle, I wanted to confirm the nature of the openscad executable before calling it with any real files, but perhaps that has caused more trouble than it's been worth.

I'll see what I can do about considering a valid executable to be one which can call `openscad --version` that doesn't write to `stderr` (such as calling a non-existent file).

fenhl, 2024-06-08:
I have reported the missing version output as a bug to Scoop: https://github.com/ScoopInstaller/Extras/issues/13425

alok-mishra, 2024-10-13:
I got openscad directly from https://files.openscad.org/OpenSCAD-2021.01-x86-64.zip  from the link on this page:
https://openscad.org/downloads.html   (not the installer)

I put the path from the gui settings, and the json shows:   "openscad.launchPath": "C:\\Apps\\openscad\\openscad.exe"

It still doesn't launch.  I can see the version number when I type --version in the command line.

I also tried both 64 and 32 bit versions.

alok-mishra, 2024-10-22:
@Antyos I set the logLevel to DEBUG, but not sure where I should see the output of it. In the Developer Tools, I do see the following error regardless of if that flag is set or not:

`
  ERR Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"
: Error: Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"

	at genericNodeError (node:internal/errors:984:15)
	at wrappedFn (node:internal/errors:538:14)
	at ChildProcess.exithandler (node:child_process:423:12)
	at ChildProcess.emit (node:events:519:28)
	at ChildProcess.emit (node:domain:488:12)
	at maybeClose (node:internal/child_process:1105:16)
	at ChildProcess._handle.onexit (node:internal/child_process:305:5)
	
	console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"
 (at console.<anonymous> (file:///c:/Users/user/scoop/apps/vscode-insiders/1.94.0-1727327122221/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:158:10032))
`
openscad.exe isn't behind a symlink (although vscode-insiders is).   And of course this is the zip version. If I do this in a terminal I get version info just fine.
`
❯ ./openscad.exe --version
OpenSCAD version 2021.01
`

If you prefer, I can also try to look at the issue (and send a PR if able).

Antyos, 2024-10-23:
@alok-mishra This looks to be an issue relating to symbolic links which is separate from the issue detailed here (which deals with the output of `openscad --version` not being properly received by the extension). I'm going to open another issue to discuss this.

bluewingtitan, 2025-01-28:
I am getting the same error, but probably for different reasons.

I am using a pretty fresh install of vscode (only platformio and C++ Extension Pack installed) and a fresh install of openscad with the install directory added to my Path. I am on Windows. I have restarted both my PC and VSCode multiple times now. I can not get this plugin to work, neither resolving openscad via my path nor using the full path.
There is no link involved in me calling openscad either way.

[image]

Antyos, 2025-01-29:
@bluewingtitan that's very odd. I'm not entirely sure what the issue is. 

I did make some fixes in another issue (I really need to just push out the update, it was like 3 months ago). Try this .vsix and let me know if it fixes the issue: [openscad-vsix.zip](https://github.com/user-attachments/files/17535419/openscad-vsix.zip).

If it doesn't, we can open a new issue for it. This update also had better logging, so there should be some extra information in the openscad output channel that will be useful in troubleshooting the issue.

bluewingtitan, 2025-01-30:
@Antyos Thank you for answering so quickly - I tried out your updated version and it worked flawlessly so far! Not sure what it was, but it seems like it was one of the things you fixed.

Antyos, 2025-01-30:
@bluewingtitan Awesome. I'll try to push out an update today (and not another 3 months from now).

Antyos, 2025-02-03:
I'm going to mark this issue as closed now. [v1.3.2](https://github.com/Antyos/vscode-openscad/releases/tag/v1.3.2) introduces the `openscad.experimental.skipLaunchPathValidation` configuration which can be enabled to bypass the version check that was causing this issue in the first place. I know it's not the optimal solution, but at least it's something. If this continues to be an issue in the future or a better solution is suggested, we can always re-open the issue.

closed 2025-02-03 by Antyos