Antyos/vscode-openscad #68: Launching OpenSCAD fails with symlink in launch path.

Antyos, 2024-10-23:
@Antyos I set the logLevel to DEBUG, but not sure where I should see the output of it. In the Developer Tools, I do see the following error regardless of if that flag is set or not:

```
ERR Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"
: Error: Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"

	at genericNodeError (node:internal/errors:984:15)
	at wrappedFn (node:internal/errors:538:14)
	at ChildProcess.exithandler (node:child_process:423:12)
	at ChildProcess.emit (node:events:519:28)
	at ChildProcess.emit (node:domain:488:12)
	at maybeClose (node:internal/child_process:1105:16)
	at ChildProcess._handle.onexit (node:internal/child_process:305:5)

console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: Command failed: C:\Apps\openscad\openscad.exe --version
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: The request is not supported: "C:"
 (at console.<anonymous> (file:///c:/Users/user/scoop/apps/vscode-insiders/1.94.0-1727327122221/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:158:10032))
```

openscad.exe isn't behind a symlink (although vscode-insiders is).   And of course this is the zip version. If I do this in a terminal I get version info just fine.

```
C:\Apps\openscad>openscad --version
OpenSCAD version 2021.01
```

If you prefer (and not complicated), I can also try to look at the issue (and send a PR if able).

_Originally posted by @alok-mishra in https://github.com/Antyos/vscode-openscad/issues/62#issuecomment-2429656689_

Antyos, 2024-10-23:
I missed your middle comment (although I do still suspect that it is separate from #62) about OpenSCAD not being a symlink. I did reproduce an issue with OpenSCAD not loading properly if it is behind a symlink, so at least that is fixed.

As it seems the issue is with VS Code itself being behind a symlink, I'm not sure how to handle that. I would be curious if you still get the issue if you run it from the resolved path.

---

I've been taking most of my code inspiration from [prettier/prettier-vscode](https://github.com/prettier/prettier-vscode) or [vscode-neovim/vscode-neovim](https://github.com/vscode-neovim/vscode-neovim), as they are relatively popular extensions which call external processes.

Looking at how something similar is done in vscode-neovim, we can see that they [create child processes with `spawn()`](https://github.com/vscode-neovim/vscode-neovim/blob/c3e8387a9e9f923ec39c28af6cb8985ca431e438/src/main_controller.ts#L67-L81). Their use case is slightly different here, as they continue streaming data in and out of the neovim instance which does not occur with OpenSCAD, but it still may give a clue.

It is possible that the call to `execFile()` should be `exec()` or `spawn()`. I probably used `execFile` at the time because it was easier, but switching to `exec()` might be a good first attempt.

https://github.com/Antyos/vscode-openscad/blob/fb74db73edf6a1b35b25c9f348e5d97319df1090/src/preview/openscad-exe.ts#L43-L46

https://github.com/Antyos/vscode-openscad/blob/fb74db73edf6a1b35b25c9f348e5d97319df1090/src/preview/preview.ts#L40-L69

These are things I will need to play around with some.

alok-mishra, 2024-10-23:
@Antyos Thanks for looking into this. I use Neovim and prettier also with the same configuration for vscode and they work well.

Wow you were right.. I opened up vscode directly from `C:\Users\user\scoop\apps\vscode-insiders\1.94.0-1727327122221\Code - Insiders.exe` (and not from the shim), and with the configuration I had mentioned, it opened up perfectly.

I also tried something else, I downloaded the app using scoop.. which puts it behind a symlink.   And I changed the config to this:
` "openscad.launchPath": "openscad.exe"`  since it's already in the path, but that also still works.

It launches even if `openscad --version` in a terminal doesn't give any output.. which is using the shim.

So basically, yes you're correct, the issue has something to do with vscode itself being behind a symlink and nothing to do with scoop paths.

Antyos, 2024-10-27:
I'm not sure why exactly why this works, but I think I fixed it. Basically, I had to specify the current working directory when calling the exe from the VS Code with a junction in the path. I'm not sure why this isn't an issue with the Neovim extension, but I'm glad it seems to work (I even tried using `spawn()`, but it made no difference).

Try this VSIX and let me know if it works: [openscad-vsix.zip](https://github.com/user-attachments/files/17535419/openscad-vsix.zip) (you can also build it yourself from the branch if you prefer). You'll need to manually install it from the extensions page in VS Code > 3 dots in the corner > Install from VSIX. If it works, I'll try to release it as a pre-release (in case it breaks something else).

alok-mishra, 2024-10-30:
Yup, this is working perfectly!  Thank you.   Just to verify, I tried to reinstall the one from the Marketplace after clearing cache and it stopped working.    And of course it worked again after I added the vsix one.

Antyos, 2024-10-30:
Perfect! Thanks for being so thorough in your testing!

I'll try to get a proper release with the fix published soon, but in the mean time you can work with the version above.

commit b9059dba (2024-10-27): Use cwd when calling openscad (fixes #68)
commit e0b35d78 (2025-01-30): Fix launch bug (#69)

* Better logging of detecting openscad executable

* Log extension name and version in output

* Add skipLaunchPathValidation config

* Bump webpack version

* Fix webpack package.json import using require()

* Resolve symlinks for openscad path

* Update Node, pnpm, and action versions in workflows

* Add try/catch to getOpenscadVersion()

* Improve logging in Preview constructor

* Fix uncaught exception with non-existant realpath

* Use cwd when calling openscad (fixes #68)

* Better logging in preview.ts

* Fix typos in changelog

* Bump github actions step versions

 [... cut]
closed 2025-01-30 by Antyos