Antyos/vscode-openscad #70: Cannot find the command: "openscad"

julianstirling, 2025-02-09:
Since upgrade to 1.3.2 pressing F5 now gives me

    Cannot find the command: "openscad". Make sure OpenSCAD is installed. You may need to specify the installation path under `Settings > OpenSCAD > Launch Path`

Nothing at all happens if I use the  **Skip Launch Path Validation** option.

I am using OpenSCAD 2021.01, on Linux

Note that `openscad --version` prints `OpenSCAD version 2021.01` to stderr not stdout.

julianstirling, 2025-02-09:
Also note that rolling back to 1.3.1 fixes the issue.

Antyos, 2025-04-22:
Hi @julianstirling I really thought I replied to this earlier, but I guess the comment didn't submit properly because I don't see it now. Sorry about that.

Is this still an issue? I had trouble reproducing it on my machine, but that may be because I was using WSL2.

If so, could you paste the debug log output in the version that has the issue? 

To enable debug logging by adding: `"openscad.logLevel": "DEBUG"` to your `settings.json` file. Then, paste the contents of the OpenSCAD output channel here. (The command to reveal the output channel is **openscad.showOutput** if you have trouble finding it). You may revert the settings change afterwards.

closed 2025-04-23 by julianstirling