Antyos/vscode-openscad #71: Preview directly in vscode

jesseditson, 2025-05-22:
Hey! I've been using this extension and it has greatly improved my experience with OpenSCAD - I find the main UI useful for when I'm ready to take a model to the slicer, but much prefer VSCode as an IDE.

I'm also pretty constantly trying to move things off my main work machine and into thinner clients, and VSCode server has gone a long way towards allowing me to edit models on my iPad which would be my preferred environment. Even on the big machine though, switching back and forth between apps can be a little cumbersome.

Given the work on https://github.com/seasick/openscad-web-gui, it seems fairly straightforward to run OpenSCAD via wasm or in an external process. I'm curious if you've considered bringing a simple preview (e.g. just a clone of the web-gui repo for now) directly into vscode, in a preview similar to how markdown or image support works.

I haven't done any VSCode extension development, so forgive any bad assumptions here. If my TODO list ever has an opening I'll take a look myself, but for now I figured I'd drop an issue in here to see if there is any prior art or known blockers in case I get around to it!

Antyos, 2025-05-23:
Thanks for the kind feedback! Having an OpenSCAD GUI display directly in VS Code has always been on my wishlist, but that would have required someone to port the GUI into wasm--which someone now has! I don't know when I'll get around to being able to implement such a feature, but it's nice to know that the base requirements have been fulfilled.

I'm happy to keep this open as a feature request until it can be properly added.

wquintal, 2025-08-01:
I would be interested in implementing this. I'm currently playing a lot with Copilot and OpenSCAD. Having a live preview after each prompt in one window would be awesome.

I will create a POC over the weekend as a starting point.

wquintal, 2025-08-06:
Good work, it may be worth just using the plugin you linked. Both our implementations are similar. I'll probably finish my implementation for fun.

And yeah wasm render is slower that is why for me it's more of a backup options if you don't have a recent executable of opencad.

Because i don't know if you have done it but supporting import in wasm is probably a pain ? I import DXF file in some of my projects and other scripts.

wquintal, 2025-08-06:
Yes it's using `wasm` compiler by default, this would work out of the box after plugin installation. But I added a property `    openScadPanelMode?: 'wasm' | 'native';` to select the native executable if you have it installed, it would be faster and a more current build of openscad.

The extensions send the compiled STL or 3MF file to the viewer to render with three.js. 3MF might be better for preview, with the possibility to see and hide different modules.

And yeah I'll add more to finish in a few days.

wquintal, 2025-08-07:
Thanks for the tips and good instructions for compiling the latest version. I'll try this too; I was too lazy to try compiling it.

Yeah, it can only help to have a better implementation in the end, and I always wanted to work on a VSCode extension.